liamcottle/meshtastic-map #2: Node TTL is way too long

garthvh, 2024-03-24:
Why does it store node data so long? 

Do you have an opt out process? 

Really is limited value to anything more than a day old.

b8b8, 2024-03-24:
Rolling 2 days i think would be the most beneficial to give a little buffer.

liamcottle, 2024-03-25:
> Why does it store node data so long?

I was asked to open source it, so it was open sourced as is. Improvements will come over time :)
The reason it stores data for so long, or forever in this case, is that I hadn't implemented purging/clean up yet, and wanted all data as is during local dev/testing.

> Do you have an opt out process?

Not yet, but I plan to add a dropdown or config so people can make their own mind up on what cut off is good for them in the UI.

I will probably also have a hard limit in the database, maybe 30 days or so, that way the db doesn't grow too large, but again, this could be in a config if you host the server yourself.

> Really is limited value to anything more than a day old.

Some others may disagree, but the config idea is probably the best option to keep everyone happy.

garthvh, 2024-03-25:
The concern is privacy, when storing data that long you open yourself up to the GDPR and the project gets accused of leaking personal info to your map.

sagegwatkin, 2024-03-27:
Privacy regulations such as GDPR are not actually relevant in this case as this is public data.

Meshtastic users have willingly, and by their own actions, set up a device that transmits data publicly. If those users don't want the data to be public, or wish the data to be treated in a specific manner, they should turn off their Meshtastic device and stop transmitting data publicly.

You can't broadcast data to someone without their consent, and then expect them to treat it in a certain manner. They haven't signed a privacy agreement with you.

Anyone can set up a device and record publicly available Meshtastic data. How long they store that data and what they do with it is up to them.

See [GDPR Article 9(2)(e)](https://gdpr-info.eu/art-9-gdpr/), sensitive personal data may be processed without consent if the data is manifestly made public by the data subject.

You may not see value in data older than 1 or 2 days, others may. The only way to please everyone is to provide this as a config option and you can configure it as you desire.

garthvh, 2024-03-27:
You are storing the data, I would check with a lawyer on both California and the GDPR when you are storing everything since you turned on the map.

Not one user has explicitly agreed to push their data to your map.

The project is asking you to not store data more than a couple of days.

b8b8, 2024-03-27:
There's also the difference between consenting to transmitting your data to the public in your own city where just users within say 100km can see it versus it being collected by a single person with MQTT enabled in region and making it accessible to the world. One single user at minimum shouldnt act as a gateway for the whole internet when there are benefits to making yourself visible to the your own community, whether it be helping to place your own nodes to fill in dead zones are just meeting local meshtastic users. Surely if someone hasn't flagged their packet for MQTT it shouldnt be relayed to the general internet.

liamcottle, 2024-03-27:
Good points from all of you.

I will implement this as per [my previous comment](https://github.com/liamcottle/meshtastic-map/issues/2#issuecomment-2017390747). There will be a configuration option available, and the publicly hosted version of the map will have a sane default, set to purge nodes etc that haven't been updated within that time frame.

> The project is asking you to not store data more than a couple of days.

If the Meshtastic project is officially asking me to have the public map set to a couple of days, that can be done. However, for others that are running their own MQTT server to collect node information, such as node positions, waypoints and text messages, they should be able to retain this data for as long as they like. Hence this should be a configuration option.

I could set up hundreds of nodes all over the city, see them on the map, and if one of them were to have an issue and go offline for several days, it would be near impossible to know which node went offline and where it is, if the node just disappears from the map. Having the UI show a different colour icon for nodes that aren't heard from in a while would be a good indication of "this node needs attention, and here it is on the map".

> Not one user has explicitly agreed to push their data to your map.

I think @b8b8 makes a good point on this in the quoted comment below;

> If someone hasn't flagged their packet for MQTT it shouldnt be relayed to the general internet. That would be the explicit consent.

I guess it's the same thing, some users have not explicitly agreed to have their packets collected and transmitted to the public MQTT server on the internet if MQTT uplink is not enabled on the node that transmitted it. So I would say this should be updated in the firmware as well...

Anyway, I'll update the public map to purge older records automatically :)

closed 2024-03-30 by liamcottle

garthvh, 2024-03-31:
The project is probably going to choose 24 or 48 hours.  There will be guidance coming soon.