liamcottle/meshtastic-map #7: Waypoint longitude shown wrong

CamFlyerCH, 2024-04-01:
Hello
Just a small issue in the pop-up of a waypoint: It seams that 360 is added to it. Position on the map is fine.

[image]

I grabbed the MQTT packet for that one:
{"channel":0,"from":3663091580,"hops_away":0,"id":4292587810,"payload":{"description":"Great viewpoint","expire":2147483647,"id":4292587809,"latitude_i":468602396,"locked_to":0,"longitude_i":75258635,"name":"Chutzen"},"sender":"!da56577c","timestamp":1711963987,"to":4294967295,"type":"position"}

Thanks for the great map ! I did not understand Neighbours yet.

liamcottle, 2024-04-01:
Thanks for the report! Yeah, the lat/lng is modified internally so it shows shortpath from NZ to USA. So the UI is showing the modified lat/lng instead of the original. Will fix when I get a chance :)

liamcottle, 2024-04-10:
This should be fixed in 61de8736abf29a0cdc2f22f7525f3bc68347a62f :)

closed 2024-04-10 by liamcottle