liamcottle/meshtastic-map #8: Feature Request: Show number of online local nodes from MapReport

GUVWAF, 2024-04-01:
The `MapReport` packet contains the [number of online local nodes](https://github.com/meshtastic/protobufs/blob/e6b4c590e7c489306c9c44e3ad1fcf62a3efd288/meshtastic/mqtt.proto#L105) of a certain node. Since when using map reporting nodes in your mesh that do not have MQTT uplink enabled will not be shown, it would be nice to have the MQTT gateway report how many other nodes one can expect in the area.

Really nice map BTW!

liamcottle, 2024-04-01:
Thanks!

Yes, very handy info. I had this saving to the database, but missed adding it to the map.

This is now shown under the MQTT Status in the tooltip if available. Otherwise nothing is shown to conserve space.

[image]

closed 2024-04-01 by liamcottle

GUVWAF, 2024-04-02:
Nice, thanks!