liamcottle/meshtastic-map #9: Feature request: implicit neighbors for mqtt nodes

prokrypt, 2024-04-02:
for nodes with fw 2.3+ which don't have neighborinfo enabled, implicit neighbors will be the ones with hop_limit==hop_start

liamcottle, 2024-04-10:
If you could give some extra details on this, I could potentially add another layer to the map.

I probably wouldn't want to pollute the neighbours layer directly, since these aren't from the neighbour packets.

But, if you could give a few packet examples on how I can determine this, I'll have somewhere to start.

Saving these might get annoying, and how to expire them. Currently, when I receive a neighbour info packet, the database just completely overwrites the old neighbour info on the node entry.

However, if we are determining neighbours based on various different packets over time, I wouldn't be able to just overwrite that entry in the db, and we would probably have to track these implicit neighbours in their own database table.

BillyBag2, 2024-04-10:
I don't think this is anything to do with the neighbourinfo. I think this is a separate hop count feature. I think it will only work to "gateway" nodes. However you can deduce hop count from other nodes to the gateway by subtracting hop limit from hop count.
As far as GUI is concerned I think you would add a hop count to all nodes markers that you have the information for. It is only up to 7. I think zero is direct. Perhaps add a D.
May be add lines to direct nodes.
Maybe add a "hull" line that would be a guide to the range of a gateway node.

BillyBag2, 2024-04-10:
You could add a "hull" around all hops less than 2 and all less than 3 etc, showing the ranges for 1,2,3 hops from the selected gateway. Perhaps show for all gateways if nothing selected but this would be confusing for gateway nodes that are close together.

closed 2025-04-25 by liamcottle