liamcottle/meshtastic-map #14: Route trace not being decoded correctly.

BillyBag2, 2024-04-11:
The route trace uses the packets to and from data in the header plus the payload that contains additional participants.
So a packet can be sent From: A To: Z asking for a route.
{ From:A To:Z }

Seen by B and "repeated".
{ From:A To:Z Route: [B] }

Seen by C and repeated.
{ From:A To:Z Route: [B,C] }

Seen by Z and so Z replies.
{ From:Z To:A Route: [B,C] }

Repeated without change. across the network.
{ From:Z To:A Route: [B,C] }

Seen by A. 
So the route is an outgoing route. But this is a reply so the route starts with the "To" recipient, then the payload and ended at the from recipient.
(To)A [ B , C ] (From) Z
A->B->C->Z
This is called "2 Hops"
A -hop-> B -hop-> C -Rx-> z
So a 2 hop route has a payload of two nodes and a total of four participants in the route. (Plus a gateway node, which is typically just an observer.)
 Very very confusing!
 
 I think you are completely missing the "to" field. This is usually to "all", a special node ID.

BillyBag2, 2024-04-15:
I have added a pull request to fix this. I have not added a migration solution for this change,
https://github.com/liamcottle/meshtastic-map/pull/16

liamcottle, 2024-04-16:
Thanks for the contribution! I have merged it in, and it's now live on my hosted map :)

closed 2024-04-16 by liamcottle