liamcottle/meshtastic-map #17: [Feature Request]: Ability to reset Neighbor Overlay

youcanstewit, 2024-04-22:
I had the Neighbor Info Module enabled on a node I kept in my car. Nodes found while driving are not true direct neighbors when I get home. Can we have the ability to reset or have the Neighbor Info expire?

GMart, 2024-04-23:
This doesn't sound like a Map issue. 
Currently the firmware will remove your neighbor links after 2x the neighbor refresh interval. So, if your regular interval is 15 mins, wait a little over 30 mins and the incorrect neighbor links should dissapear from the map. (If you are still connected to MQTT)

liamcottle, 2024-04-28:
Unfortunately, I don't think I would allow for resetting or deleting the neighbours on the map, as your node is probably just going to rebroadcast those same old neighbours later on, and they will reappear on the map even if you're no longer in range (since they are still in the node memory).

I'm not too knowledgable on the firmware side of how it expires neighbours internally, but based on what @GMart said, you could try adjusting your neighbour interval on the node side.

I could possibly add a new `Neighbours Max Age` setting to allow hiding neighbours that are older than x seconds, similar to how you can already hide nodes and waypoints with the `Nodes Max Age` and `Waypoints Max Age` settings.

Although the above would not actually be possible on a neighbour specific level, but would be based on the timestamp that the entire list of neighbours was received over MQTT. So if a neighbour broadcast was received by my map 10 seconds ago, but all of those neighbours were actually heard over an hour ago, and you hide neighbours older than 30 minutes, you will still see all of those neighbours that are 1 hour old, since we can only determine age based on the 10 seconds ago timestamp (of packet being received from MQTT).

I do see the firmware includes [`last_rx_time`](https://github.com/meshtastic/protobufs/blob/master/meshtastic/mesh.proto#L1534) on the `Neighbour` proto, but since this isn't sent over the mesh or to MQTT, it's not possible to filter out for each individual neighbour. Maybe this could be broadcasted over RF in a firmware update, to allow for such filtering?

closed 2024-05-08 by liamcottle