liamcottle/meshtastic-map #20: Remove extra image requests

GMart, 2024-04-23:
The site currently requests about 2 dozen device images that don't exist each time the site loads. There could be a check to make sure the image exists before trying to load the image.
Or we can add a placeholder image for each device type. 
I guess there already is one, but not sure how that is generated.

liamcottle, 2024-04-23:
Related: https://github.com/liamcottle/meshtastic-map/pull/18#issuecomment-2073593020

liamcottle, 2024-04-23:
This adds an extra 1mb or so to responses, since there's like 12k nodes in the db now lol...

https://github.com/liamcottle/meshtastic-map/tree/feature/hardware-model-image-urls

GMart, 2024-04-25:
Ok, then I guess the current way works fine. 
Could you add a server side script, then the response would only be the current device types with image urls? So maybe 30 URLs total. It would only need to update maybe once per day.
I'm not up to date with how scripting works these days.

closed 2025-04-25 by liamcottle