liamcottle/meshtastic-map #29: UX - Splt stacked nodes

Talie5in, 2024-05-17:
When nodes are stacked on each other, as noted it's hard to click in to see a node beneath to see details, could calculate if they are within 10meters of each other for example?

Can tell there is more then one node here due to having "Heard" enabled on another node and you still have a blue line next to it.

[image]

An another mapping/visual tool I use - https://lukeprior.github.io/nbn-upgrade-map/?suburb=mount-barker&state=SA&commit=main

When there's more then one location there it displays a number on the dot, and when click spreads them out and can click on the individual marker for details.
[image]
[image]

Talie5in, 2024-05-17:
Just realised you do this in further out zooms - just not when down close.

[image]
[image]

liamcottle, 2024-05-17:
Yeah, this is the marker cluster plugin. The best fix is probably to adjust the configured zoom level for it, so it still clusters when zoomed in further.

hassos246, 2024-10-06:
Except that wouldn't work if the nodes have the same location, witch they often get with the approximated position.

closed 2025-04-25 by liamcottle