liamcottle/meshtastic-map #35: Some TextMessages are too long to store in the database

fifieldt, 2024-05-22:
When running meshtastic-map for long enough, some exceptions like below arise.

`Invalid `prisma.textMessage.create()` invocation in
meshtastic-map/src/mqtt.js:312:42

  309 }
  310 
  311 try {
→ 312     await prisma.textMessage.create(
The provided value for the column is too long for the column's type. Column: text
    at In.handleRequestError (meshtastic-map/node_modules/@prisma/client/runtime/library.js:122:6854)
    at In.handleAndLogRequestError (meshtastic-map/node_modules/@prisma/client/runtime/library.js:122:6188)
    at In.request (meshtastic-map/node_modules/@prisma/client/runtime/library.js:122:5896)
    at async l (meshtastic-map/node_modules/@prisma/client/runtime/library.js:127:10871)
    at async MqttClient.<anonymous> (meshtastic-map/src/mqtt.js:312:17) {
  code: 'P2000',
  clientVersion: '5.11.0',
  meta: { modelName: 'TextMessage', column_name: 'text' }
`

liamcottle, 2024-06-06:
I've seen these logs, but haven't investigated them any further.

My initial thoughts without actually debugging it, are that the packets are being decrypted and successfully parsed as a Data message (in protobuf) even though the decryption key was not the correct key.

And by chance, the portnum just happens to line up with `TEXT_MESSAGE_APP` and the decoded text is just full of garbage.

fifieldt, 2024-06-06:
Great analysis. I mainly added this as a placeholder to investigate.

closed 2025-04-25 by liamcottle