liamcottle/meshtastic-map #36: Unable to decrypt AES256 key

rohanki, 2024-05-26:
If using a different randomally generated PSK for primary channel, meshtastic will generate an AES256/32 byte key. 
mqtt.js is currently unable to decrypt this packet even when passed the key as an argument. I've added some debugging in - receiving a packet with an AES256 key throws the following:

RangeError: Invalid key length
    at Decipheriv.createCipherBase (node:internal/crypto/cipher:121:19)
    at Decipheriv.createCipherWithIV (node:internal/crypto/cipher:140:3)
    at new Decipheriv (node:internal/crypto/cipher:289:3)
    at Object.createDecipheriv (node:crypto:155:10)
    at decrypt (/home/ro/meshtastic-map/src/mqtt.js:241:37)
    at MqttClient.<anonymous> (/home/ro/meshtastic-map/src/mqtt.js:331:29) {
  code: 'ERR_CRYPTO_INVALID_KEYLEN'
}

I'm not overly experienced with javascript, but I was able to solve the problem by adding the following:

liamcottle, 2024-05-26:
Thanks for logging this. I had hardcoded the key type. Will get a patch pushed out for this a bit later tonight.

liamcottle, 2024-05-26:
Fixed in 43d6f5be75b7b414653517dedc64402dfc80fc55. Feel free to reopen if you still have this issue :)

closed 2024-05-26 by liamcottle