liamcottle/meshtastic-map #40: No packets seen on MQTT

cyberorg, 2024-06-03:
None of the nodes show mqtt topic they are seen on, 
see this for example: https://meshmap.net/#3662844492

Last seen	via	on root topic
5min ago 	self 	msh/IN/Bir/Alft
5min ago 	[!da529480] msh/IN/Bir/BU

The same node:
https://meshtastic.liamcottle.net/?node_id=3662844492

Shows:
No packets seen on MQTT

ianatkin80, 2024-06-04:
Also observed here - it looks like MQTT *might* be connected again but neighbour info is not updating?

liamcottle, 2024-06-06:
> No packets seen on MQTT

This feature relies on the server collecting all service envelopes. At this time, this feature is disabled on my public map. The UI should actually hide this section entirely in this case, but I had not got that far...

Also, it was implemented in the inverse to the meshmap site. My map would show the topics your own node was publishing to, not the topics that other nodes published your packets to.

This is probably a feature that should be revisited, but would require more tracking/tables in the database to make it work without having to store all of the received service envelopes.

closed 2025-04-25 by liamcottle