liamcottle/meshtastic-map #49: neighbour information not updated/missing

jranma, 2024-07-24:
Hi
Looks like the neighbour info (heard us / we heard) has not been updated for several days. 
Is it disabled now ? 
Will you consider enabling again ? Very usefull feature, on my point of view.

liamcottle, 2024-07-24:
Hi, this was disabled on the official MQTT server by the Meshtastic developers. You could run your own server to get around this new limitation.

jranma, 2024-07-24:
Thanks. Can you be a little more specific? What has been deactivated, and why?
I don't understand what the module in the firmware is for, if the data can't be retrieved?

liamcottle, 2024-07-24:
I don't have a link to the discussion on me right now, but in the official Meshtastic Discord server you should be able to find more about it in the `#mqtt` channel.

The neighbour info is still useful and can be collected by running your own node that is connected to your own MQTT server, it's just no longer broadcasted by the public Meshtastic MQTT server.

This will of course only allow you to collect neighbour info from your area, and not the entire world.

I believe this was being done to minimise the amount of traffic being sent/received.

jranma, 2024-07-24:
Ok, thks for the infos. 
I will try to run your map software on my own MQTT server, then.

closed 2024-07-24 by jranma