liamcottle/meshtastic-map #50: Another way to determine MQTT as Online/Offiline (meshtastic/firmware/pull/4305)

Talie5in, 2024-07-29:
https://github.com/meshtastic/firmware/pull/4305 has changed/removed the status topic from MQTT which now stops the ability to see if the MQTT is Online or Offline in the dashboards.

Is there another method that can be used to determine the status?

liamcottle, 2024-07-29:
Not sure on that one...

Wasn't expecting that change. I guess you'll just have to hope that your node is correctly connecting to your MQTT server, and just assume that it's always connected.

Looking at the change, they intend to publish the node info on each new connection to the MQTT server, although this is not implemented yet. This still wouldn't tell you if it's currently connected, just that it had connected at a specific timestamp.

As far as I know, you will no longer be able to determine if you node is _currently_ connected to the MQTT server. Maybe someone else knows a way to tell?

liamcottle, 2024-09-02:
Closing out, as this feature has been updated to be based on when a packet was heard being gated to MQTT by the node in question.

closed 2024-09-02 by liamcottle