liamcottle/meshtastic-map #53: Project rewrite

KomelT, 2024-08-04:
Hello, I am trying to contribute features to this project, but the Vue app, mqtt.js, and index.js are getting bigger and harder to read and understand. It would be smart to rewrite the front-end into a proper Vue app and use typescript for every aspect.

I am willing to help.

liamcottle, 2024-08-06:
Hi there, yes, this is in the todo list. I will likely use vite and vue components. I am working on other projects at the moment, and I'm doing the same conversion...

As much as it would be great to have someone else do all of that for me, I would prefer to do the conversion myself, otherwise I'll have no idea how my own codebase works anymore lol.

I appreciate that you want more features, and also appreciate you're willing to help! If I can't get to this fast enough, you are of course welcome to use your own fork in the meantime.

Ideally it won't be too long, I'm just prioritising other work at the moment. I probably wouldn't accept a PR for a rewrite though sorry :)

KomelT, 2024-08-06:
I understand. Thank you for explaining. 😊

closed 2025-04-25 by liamcottle