liamcottle/meshtastic-map #60: MQTT excluded portnums

KomelT, 2024-08-19:
There were added some Public MQTT server enhancement.

Original post: https://discord.com/channels/867578229534359593/871558934916390953/1260642727335428176

TLDR, these are only allowed portnums on Meshtastic MQTT server:
- NodeinfoApp
- TextMessageCompressedApp
- TextMessageApp
- PositionApp
- TelemetryApp
- MapReportApp
- RoutingApp

liamcottle, 2024-08-19:
[image]

liamcottle, 2024-08-19:
I could probably add a note to the readme, and also note that running your own MQTT server doesn't have these restrictions.

closed 2025-04-25 by liamcottle