liamcottle/meshtastic-map #63: Feature-Request: Precision circles around nodes

CamFlyerCH, 2024-08-25:
Not urgent and you have probably a fair amount of work and changes about the map anyway:

It would be nice to have circles around nodes or at least a visible indication in the node popup about the precision setting the node have when sending the coordinates. (Like the apps now do and other maps).

liamcottle, 2024-08-26:
Hey!

I've pushed out the first iteration of this. The info tooltip for a node will show the position precision in meters/kilometers, and when you click a node it will show a blue circle with that distance, if it's available.

Let me know if you have any feedback, will close this as completed.

closed 2024-08-26 by liamcottle

CamFlyerCH, 2024-08-26:
Wooow , that was quick. Results are a bit inconsistent, also compared to meshmap.net. There are nodes were it works perfectly. (Example: https://meshmap.app/?node_id=3664082500). Other not so much. (Example https://meshmap.app/?node_id=1236633468) 
[image]
[image]
Did you implement all levels of precision ?

liamcottle, 2024-08-27:
Yep, all of the relevant ones were implemented: c080acecf9ed3231d88625d670c3f40a1aaaa8d4

https://meshtastic.liamcottle.net/api/v1/nodes/3664082500
- Looking at the data collected by my map, I can see the first node does in fact have a `position_precision` of `14`.
- Since that value is available, the ring is shown.

https://meshtastic.liamcottle.net/api/v1/nodes/1236633468
- You said this node is showing a ring on meshmap.net but not on my map.
- According to my API, I have not collected a `position_precision` for this node.
- Are you sure it's being sent by this node, and isn't stale data on the other site?

CamFlyerCH, 2024-08-29:
So I tried different configurations of this node. Nothing helps. Other nodes work fine, sometimes the position is updated, but the precision not, but later then yes. 

I send the MQTT data from my receiving node to my MQTT broker , that then bridge it to yours and to the official one (there I need to move the topic from msh/CH to msh/EU_868, since the CH topic is not received/monitored there.

In JSON I receive that data:
{"channel":0,"from":1236633468,"id":1523123059,"payload":{"altitude":525,"latitude_i":468451328,"longitude_i":75759616,"precision_bits":13},"rssi":-30,"sender":"!da56577c","snr":5.75,"timestamp":1724962755,"to":3663091580,"type":"position"}

Encrypted it's not readable. 

And the same data is received on https://meshmap.net/#1236633468 without issues. It's a strange issue.