liamcottle/meshtastic-map #68: [Feature Request] Show connections on map

DL4OCE, 2024-08-29:
This might increase people's motivation to get in touch with each other: is it possible to extend the map by lines between nodes that can directly hear each other?

I could also think of arrows showing the diretion of the communication:

Node_A --> Node_B: only Node B can hear Node_A
Node_A <-- Node_B: only Node A can hear Node_B
Node_A <--> Node_B: Node_B and Node_A can hear each other

Line thickness or color (grey tones) could represent the quality of the connection either by signal strength and/or last time heard.

lagunacomputer, 2024-09-01:
I can't get any "Nodes i heard" or 'Nodes that heard me" lines for my node on the map.
Can someone layout EXACTLY what the requirements or settings are to do get the links to show up?
I have:
-MQTT setup to liamcottle per the instructions on the map website, green and online on map
-Uplink enabled in primary Long_Fast channel
-Sent and received successful Long_Fast messages and tracerts

GMart, 2024-09-02:
This is already shown on the map. You need to have uplink on a public channel, and turn on the neighbors module on all nodes. Set it to something like 1000 seconds. Then on the map, turn on the neighbors layer. You will see who heard what node, and if they can both be heard, the line is thicker.

DL4OCE, 2024-09-02:
Okay, tbh, I didn't know that.
But.. I think that the way connections are shown at the moment is pretty useless as it doesn't only show the nodes that directly hear each other via LoRa, but also the ones being connected over the internet.

What I would like to have is a view on the structure of the mesh.

DL4OCE, 2024-09-02:
> I can't get any "Nodes i heard" or 'Nodes that heard me" lines for my node on the map. Can someone layout EXACTLY what the requirements or settings are to do get the links to show up? I have: -Tried firmwares 2.3, 2.4 and am currently on the latest 2.5 -MQTT setup to liamcottle per the instructions on the map website, green and online on map -Uplink enabled in primary Long_Fast channel -Sent and received successful Long_Fast messages and tracerts Note: I only have one node close by that I can hop thru to reach other nodes (all my traffic must go thu it to get to other nodes), and it is NOT connected to MQTT. But other farther nodes that should hear that in between repeater-client are Green and Online on the map

Same here. Even when it shows blue lines between the nodes, I won't get any entries in the heard lists..

DL4OCE, 2024-09-02:
> This is already shown on the map. You need to have uplink on a public channel, and turn on the neighbors module on all nodes. Set it to something like 1000 seconds. Then on the map, turn on the neighbors layer. You will see who heard what node, and if they can both be heard, the line is thicker.

Okay, tbh, I didn't know that.
But.. I think that the way connections are shown at the moment is pretty useless as it doesn't only show the nodes that directly hear each other via LoRa, but also the ones being connected over the internet.

What I would like to have is a view on the structure of the mesh.

closed 2025-04-25 by liamcottle