liamcottle/meshtastic-map #71: Code Update - Nodes on Map Disconnected

skykingisepic, 2024-09-14:
Just did a repo update, had been running for 4 months

Now my main node is showing Disconnected when it isn't.

Are there startup options required besides --mqtt-broker-url mqtt://myserver:8183 --mqtt-username MeshEast --mqtt-password private --mqtt-topic msg/TN --purge-interval-seconds 43200 --purge-nodes-unheard-for-seconds 172800

Nodes use same login/topic info for MQTT server and were reporting Online before the update

logdog16, 2024-09-19:
You're probably missing a command line option or two after [2cd6e57](https://github.com/liamcottle/meshtastic-map/commit/2cd6e570eb899ebd5d16e8fa6ff94965288c719e) merged and made all options off by defaut. Try adding `--collect-positions` and/or `--collect-waypoints` to your args

skykingisepic, 2024-09-19:
Changed startup to:

node src/mqtt.js --mqtt-broker-url mqtt://epicnet.us:8183 --mqtt-username MeshEastTN --mqtt-password privcom4us --mqtt-topic msg/TN --collect-service-envelopes --collect-neighbour-info --collect-map-reports --collect-positions --collect-waypoints

Added two collect and dropped the two purge

Still no joy - empty data sets in mariadb.

logdog16, 2024-09-19:
Ahh, I think I misunderstood your original issue. You got me curious, so I spun up a new VM to test and can repro similar behavior (connected to MQTT but nothing written to DB) when using `docker compose up` however, when I run the same command without using docker `node mqtt.js <Mqtt details omitted> --collect-positions --collect-neighbour-info --collect-map-reports` it works for me.

I'll let you know if I manage to find anything or get my test setup working with a bit more hacking