liamcottle/meshtastic-map #72: Research: Multiple nodes with the same ID !6e2718a

CamFlyerCH, 2024-09-15:
By accident I found a the node ID !6e2718a that probably is set on my many Meshtastic nodes. It is always a RAK4631. The battery state fluctuates a lot between different values. 

[image]

This is NOT an issue with this map ! Also the Meshmap indicates the issue my showing MQTT nodes receiving this ID from different continents in a short time :
[image]

Also visible in the position history:
[image]

I was thinking you could probably extract more data from nodeinfo packets you MQTT server received to understand this issue better. I don't know how the ID is exactly generated and how one affected user can change it. I hope this is interesting for you too. I wonder if there are other IDs affected by this. I did not find the official documentation for the way the ID is generated. According to different LLMs the BT Mac address is used , but probably also other things.

Forge36, 2024-09-15:
Applicate of #47 ?

liamcottle, 2024-09-16:
Some messages related to this in the Meshtastic Discord:

- https://discord.com/channels/867578229534359593/871553024030556250/1279051666246864896
- https://discord.com/channels/867578229534359593/871553024030556250/1267431315817500692

> I don't know how the ID is exactly generated and how one affected user can change it.

As far as I know, the solution is to factory reset the device, in which case it will then re-generate the node ID based on the actual hardware BLE mac address.

I had a couple of "new" nodes that had weird IDs, and factory resetting them did in fact resolve the issue.

[image]

CamFlyerCH, 2024-09-16:
OK, thanks for that. I was not able to find these. My bad or the fault of duckduckgo. In this case all is clear now. I hope the affected people have no nodes in range with the same ID or find the solution as mentioned on Discord.

closed 2024-09-16 by CamFlyerCH