liamcottle/meshtastic-map #73: [Privacy] Obfuscate or filter out node info that includes any precise location

ariveramedentem, 2024-09-21:
Today, nodes are reported on the map with precise location. Newcomers who are not aware or familiar with how location is reported (particularly on older firmware) will be leaking their precise location.

Two options to resolve:

1. Ideal Scenario - location is auto obfuscated on the way in. Then we can still populate the map but mitigate the risk.
2. Less Ideal Scenario - packets with precise location are dropped.

ariveramedentem, 2024-09-21:
PR opened here: https://github.com/liamcottle/meshtastic-map/pull/74

GMart, 2024-09-23:
I don't think this is necessary. I like having this map as an option with the location history, and also, at least for the US, nobody is going to upload to this map unless they know about it, as the public meshtastic.org server is not connected. 
Also, newer versions of the firmware are opt-in for MQTT. If that flag is added, this won't be necessary. 

If it must be done, then I would like the limit to be 16 or 18 bits of precision (About 100m)

ariveramedentem, 2024-09-23:
In the PR its behind a flag now, and defaulted to false.

ariveramedentem, 2024-09-23:
Also added configurable precision.

closed 2024-09-30 by medentem