liamcottle/meshtastic-map #77: Date format in a detail window

Coopersmith-24601, 2024-10-03:
I believe this may be incorrect (for US dates anyway).
[image]

liamcottle, 2024-10-03:
Hey, dates are formatted as `day/month/year`. It's the less confusing format ;) I could probably change to `01/Jan/2024` instead of using numeric month, otherwise the other format that's generally used is `year-month-day`.

Coopersmith-24601, 2024-10-04:
Fair, completely fair.  If so though then there is a discrepancy in the "From" and "To" section of the time period selection.  ;)

liamcottle, 2024-10-05:
Ohhh, true... I see that in your screenshot now :D I guess that's the browser putting it in that format. I'll have to investigate if this can be changed.

closed 2025-04-25 by liamcottle