liamcottle/meshtastic-map #79: Feature Request - Add date stamp on gated messages

Paul-Pressure, 2024-11-02:
Feature Request

We are using the gated message feature to check message arrival over Lora radio and its extremely useful. 

Please add, date-stamp to each message. 

Love your work is a fantastic map and features

SubnetConflict, 2024-11-02:
^ ^ One up on this one please - Especially when monitoring several nodes

liamcottle, 2024-11-07:
Howdy! Thanks for the feedback :)

I've updated the messages UI to show the full date time instead of the relative time ago when you click on a message in the list.

Hope this helps!

[image]

closed 2024-11-07 by liamcottle

Paul-Pressure, 2024-11-07:
Hi Liam

Thank you.  to explain what we are using your gated messages for see  https://pole1.co.uk/trial/
this is extremely useful data so our local mesh can see who receives a message sent via mesh/lora

I was hoping for each individual message to have a time-stamp so we can see when it received via your MQTT.

Hope that makes sense 🙂

Cheers
Paul

Screenshot.2024-11-07.at.11.15.03.PM.png (view on web)<https://github.com/user-attachments/assets/ff133c8d-4e49-4a12-92fc-605b28422901>